metroid-maniac/SCFW #1: White screen after flashing firmware

Gamemaster2022, 2024-03-22:
Hello, I downloaded your custom firmware and put it on my SD card as said in the description. Then I tried the kernel.gba - it worked I got into the menu. I liked the new menu so I decided to go ahead and flash the cartridge with the firmware.frm. Everything finished successfuly but after restart I got only white screen and nothing else. I am worried I bricked it. Any suggestions what to do now? 

Help is greatly appreciated.

Thanks a lot

metroid-maniac, 2024-03-22:
Hi.

Could you describe what happens when you try to use the cart? Do you see the Game Boy logo and a normal Nintendo logo before going to the white screen?

If you have a Nintendo DS and a DS flashcart, there are a few ways to unbrick the cart. You can try running flashmp or scflash_sd.  If your DS flashcart runs YSMenu or TwilightMenu++, you can also run sd_v185 natively to use the official firmware updater too.
[flashmp.zip](https://github.com/metroid-maniac/SCFW/files/14723755/flashmp.zip)
[sd_v185.zip](https://github.com/metroid-maniac/SCFW/files/14723756/sd_v185.zip)
[SCflash_sd.zip](https://github.com/metroid-maniac/SCFW/files/14723757/SCflash_sd.zip)

Gamemaster2022, 2024-03-22:
> Hi.
>
> Could you describe what happens when you try to use the cart? Do you see the Game Boy logo and a normal Nintendo logo before going to the white screen?
>
> If you have a Nintendo DS and a DS [...]

Thanks for quick reply,
basically I just insert the cart in with my pre-formatted 2GB SD. GAME BOY logo with Nintendo appears and then it shifts into pure white screen (see attached photos). I unfortuanetly don't own a DS so that will be a problem

[image]

[image]

Gamemaster2022, 2024-03-22:
Is there any way to unbrick it without DS?

metroid-maniac, 2024-03-22:
Do you have any other GBA flashcarts?

Gamemaster2022, 2024-03-22:
> Do you have any other GBA flashcarts?

I have only Evedrive GB X5 that doesn't run any GBA stuff as it's GB/GBC only