metroid-maniac/SCFW #3: "Could not overclock EWRAM" on GBA SP after soft-reset.

Janni9009, 2024-04-01:
The kernel successfully overclocks EWRAM on coldboot, but fails to do so again when soft-resetting.
SCFW is flashed as firmware, firmware is version 0.4.4, kernel is version 0.5.0, console is a EUR GBA-SP, cart is an SC-µSD (clone).

metroid-maniac, 2024-04-01:
Hi @Janni9009 
The commit 6e53175 should fix this.
Can you please test this kernel?
[scfw.zip](https://github.com/metroid-maniac/SCFW/files/14823606/scfw.zip)
If you can confirm this bug is fixed, I will make this the next release.

coreflake1, 2024-04-01:
Had the same error on my end (now that I finally got everything running). 
Tried the new version, now it's always able to overclock EWRAM.

Janni9009, 2024-04-01:
Can confirm, it's fixed. @metroid-maniac

closed 2024-04-01 by Janni9009