metroid-maniac/SCFW #6: Bugs on various consoles

Veund, 2024-05-14:
So I downloaded this and installed it onto my Supercard SD

When run on the GBA Player GameCube using the official disc PAL I get this error.
[image]

When I play it on the original GBA I get this error repeatedly then just a white screen at the end.
[image]

The Original NDS is not able to overclock the EWRAM, nor is the Gameboy Micro nor the NDS lite .
[image]

The only device that seems to get it all right is the Gba SP i tested a 001 but a 101 might give a different result as there is CPU changes between the two.

This is al tested on 0.5.2 on the Supercard SD Micro SD version with a 32GB SD card, I also have a Mini SD version that I can test things on and essentially every Gameboy advance compatible hardware excluding a GBA SP 101.

Veund, 2024-05-14:
Found more things to test.

On this GBA it boots fine but when launching a game it is stuck on a white screen, no EWRAM overclock.
[image]

On the Analogue Pocket it boots up fine but says invalid header when game is launched, oddly enough the Pocket also allows EWRAM to overclock.
[image]

OmDRetro, 2024-07-08:
## Summary:
* GBA Player GameCube, using the official disc for PAL region, cannot find directories.
* An `AGS-001` model GBA experiencing "SDRAM write failed" errors
* Unable to overclock EWRAM on NDS/NDSL & GB Micro
* Unable to boot cartridge on an FPGA clone

**Your report lacks detail and would require a deeper analysis**

***
Possible Workarounds / Analysis

* GBA Player
    * Unable to find directories could mean one of the following:
        * `x:/scfw` folder does not exist. ~ Try placing in directories
        * Cannot read MicroSD ~ Try using different SD cards or filesystems
        * Have you tried booting the SuperCard via PicoBoot like the [one shown here](https://youtu.be/MLI9KaToA4g?si=hYEFP3mCULK2DoVl&t=3143)?
        * The `Gameboy Player` has [some form of DRM](https://gaming.stackexchange.com/questions/125711/are-there-any-gba-games-that-the-gamecubes-game-boy-player-cannot-play-well) that prevents booting of ROMs
* AGS-001 having SDRAM write failed errors
    * You have not indicated how to replicate this problem thus it will be considered invalid until I can replicate this on my AGS-001
* Unable to overclock EWRAM on NDS/NDSL & GB Micro ~ `invalid concern` as of now
    * You have not read the changelog since [SCFW v0.50](https://github.com/metroid-maniac/SCFW/releases/tag/0.50)
* Unable to boot cartridge on an FPGA clone
    * Can you check if stock firmware SuperCard SD's work on that clone including playing GBA ROMs on it?

Veund, 2024-07-08:
Sorry each one of these happens on boot of the same scupercard set up the exact same way

I have no modded GameCube and should NOT require a modded one for this to function as this is the intended use case of the Gameboy player. using homebrew should be immaterial. yes all folders were present, micro SD works on all other devices this exclusive to the gameboy player.

the AGB-001 errors on boot, there is nothing that can be done because on that specific GBA it errors before menu. but as you can see in a later photo it works on another GBA no issues let me know if you want more details.

I was just trying it on ever device I own which is a lot of different devices and revisions.

yes a stock supercard does indeed work in the Aalogue pocket and plays games with no issues. I tested it on my second supercard with stock firmware

OmDRetro, 2024-07-10:
We can start with the AGB-001 that can boot SCFW. You mentioned that no ROMs are booting, correct?

- What is the ROM you're trying to load specifically? You observed a white screen on what ROM? Be specific.

Veund, 2024-07-10:
No sorry you miss understand, that happens before it gets to the scfw menu,
i turn it on it starts spitting out that error then goes to a white screen
and hangs

OmDRetro, 2024-07-10:
You've shown me two AGB-001's.

One that can properly boot but no ROM loads and another that has an SDRAM issue which can indicate a problem with the GBA's edge connectors not having proper contacts with the SuperCard SD pins.

I'm talking about the GBA you used that can navigate SCFW(second picture). Can you provide a name of a ROM or patched ROM that you're trying to run that ends up with a white screen?

Veund, 2024-07-20:
Sorry i misunderstood, on every device pokemon emerald was tested and works on the other consoles