metroid-maniac/SCFW #7: Modded GBA Problem

Bizzy-Shz, 2024-05-16:
The Games won't start on my GBA with IPS Mod. On the Original GBA SP and NDS Lite are no problems.

https://github.com/metroid-maniac/SCFW/assets/110717923/dad2c110-8487-4b16-b05e-f366716b5b88

https://github.com/metroid-maniac/SCFW/assets/110717923/a59143dd-8380-40e3-b942-8a7b351f1d32

https://github.com/metroid-maniac/SCFW/assets/110717923/a8356280-3635-4b39-8634-5a7596b4600e

Bizzy-Shz, 2024-05-18:
I tried patched Roms, unpatched Roms, unpatched Roms with patch sav sci, with pre patch and without.

The Hardware from SC and GBA is ok.
Hope anything from this info helps your progress.

Here some Picture from IPS PCB and SC.
[image]
[image]
[image]

Bizzy-Shz, 2024-05-21:
I switched back to the OFW with your OFW.frm and now my modded GBA can play the Games again.

Hope your next update helps, it's a great project

OmDRetro, 2024-07-06:
This might be related to the EWRAM overclock since it works on my clone(EXEQ Game Box). I'll be making a build later to test this out

OmDRetro, 2024-07-08:
## **NOTE:**

Some ROMs, oddly, do not function on vanilla AGB-001's. It doesn't matter if the AGB-001 is modded or not. Some ROMs don't function on it as I'm able to replicate this on my end.

### `I've observed better compatibility found on GBA SP models(incl. clones like the EXEQ Game Box SP) as well as the DS/DSL.`

fernandobonfimandrade, 2024-08-14:
@Bizzy-Shz  You said you switched to OFW.frm, does that mean you were able to go back to the original supercard firmware? Do you have it to share?

OmDRetro, 2024-08-23:
I have done multiple tests and two processes may be required here.
This is also a partial duplicate of another concern.

- Disabling the [Increase ROM Speed](https://github.com/metroid-maniac/SCFW/issues/10#issuecomment-2266148248)
    - The ROM you're trying to load has the `increase ROM speed` feature. This has to be disabled via patching
- Entering Multi-boot mode for 2 secs prior to loading the game
    - This step is optional, but some devices/SuperCard SDs having various chips require an additional 2 seconds preparation time before loading the game. (Don't know why this behavior exists for some SuperCard SDs)
        - Basically before you boot the ROM(During the boot intro), press `START` and `SELECT` simultaneously and you'll hear a short jingle along with the disappearance of the `Nintendo` logo.
        - Wait for 2 to 4 seconds before pressing any direction key or A / B keys. This will continue the boot process

The ROM should boot afterwards:

[image]

OmDRetro, 2024-09-06:
Since 2 weeks had passed and there's no response from the user, I'm closing this bug as the solution was already shared.

closed 2024-09-06 by OmDRetro