metroid-maniac/SCFW #8: pre patching

jmsunseri, 2024-05-17:
would it be possible to keep a list of games you have previously seen and when the device is turned on it will pre patch them and  you won't need to patch them when the you select the game?  or possibly patch them the first time the game is run and not needing to patch over again.

metroid-maniac, 2024-05-17:
I have planned the second option, which is to generate a patch the first time the game is run. However, the Supercard SD driver is currently extremely slow at writing. I would want to improve this driver before caching patches like this.

jmsunseri, 2024-05-17:
i have two of these cards coming in the mail.  i'm very interested to see where this goes.  patching aside it seems to take a while to load up the games.  would be nice to have a better loading screen.    i wish i could help with all this

BigBrainAFK, 2024-08-21:
> I have planned the second option, which is to generate a patch the first time the game is run. However, the Supercard SD driver is currently extremely slow at writing. I would want to improve this driver before caching patches like this.

I don't know why implementing patch caching would be dependent on the SD driver performance itself. i can imagine anything is better than sitting through patching all the time. The caching would also not work different depending on SD card speed I assume.

OmDRetro, 2024-08-22:
> I don't know why implementing patch caching would be dependent on the SD driver performance itself. i can imagine anything is better than sitting through patching all the time. The caching would [...]

I think you misunderstood what @metroid-maniac wrote here.
### It's more of which feature or enhancement should be prioritized first. In this case, optimizing SCFW's flashing process comes first as the rest can thought of / worked on later.
Also, there's no ETA for certain features( ___just in case you're thinking of that___ ). We're working on this in our spare time so releases will come as soon as it's ready.