metroid-maniac/SCFW #9: Minor enhancement - GOOMBA implementation

OmDRetro, 2024-05-17:
Based on [this tutorial](http://www.webpersona.com/goomba/tutorials.html), concatenating the emulator(Goomba/Goomba Color in this case) and the ROM file allows the emulator to see the game file, but the resulting stream of concatenating the two will have to be flashed.

```
//DOS shell command for concatenation of binaries
copy /b goomba.gba+game1.gb+game2.gb goombamenu.gba
```

```
//BASH command for concatenation of binaries
cat goomba.gba game1.gb game2.gb > goombamenu.gba
```

The resulting file `goombamenu.gba` is what needs to be on the flashchip / rewritable rom chip. Not sure how other flashcarts implement this, but I'm adding it here as a helpful source.

metroid-maniac, 2024-05-17:
Concatenating the emulator ROM with the game ROM is enough to make the emulator boot up the game.

I'm not quite sure how "return to menu" functionality works yet. Pogoshell seems to have defined a standard for this, which goomba and Sterophonick's Simple kernel have implemented, but it uses a lot of magic addresses so I haven't figured it out yet. I guess that can be figured out later.

OmDRetro, 2024-06-03:
Done pushing changes to new branch.
Goomba successfully implemented.

Please check the [bleeding-edge kernel here](https://github.com/metroid-maniac/SCFW/tree/OmDRetro-SCFW-Kernel)

closed 2024-06-03 by OmDRetro