metroid-maniac/SCFW #10: final fantasy series whitescreen patch

multi-vitamin, 2024-05-20:
final fantasy series currently runs on scfw with a whitescreen.

I tried patching it with the SC v2.71 program and it runs fine.
I compared it with binary viewer and there is a specific part that is patched.

I don't know exactly if this affects this series only.
This is the data that was replaced.

10 9F E5 04 12 80 E5 -> 10 9F E5 **00 00 A0 E1**
10 9F E5 04 12 80 E5 -> 10 9F E5 **C0 46** 80 E5

OmDRetro, 2024-07-06:
Hm... If you look for that specific hex pattern in a hex editor / binary viewer, the offset starts at `11A31` for `FF 1&2`.

Replacing the last 4 bytes of that pattern with `00 00 A0 E1` will allow the ROM to boot in SCFW(while keeping all kernel settings at default). This sounds similar to the issue both @metroid-maniac and I had when we attempted to run the `HVCA` emulator for the first time.

Allow me to quote him:

`"Like retail games, this emulator tries to increase the ROM speed
Which supercard is not compatible with. I just disabled that"`

Based on what the SuperCard Patcher does(v2.71), it appears to disable the `increase ROM speed` feature since the SuperCard is not compatible with that.

In the meantime, you can use that modified ROM to load FF games on SCFW 0.5.2 as this may require a kernel enhancement.

OmDRetro, 2024-08-02:
### Information carried over from [GBA Temp thread](https://gbatemp.net/threads/scfw-custom-firmware-kernel-for-supercard.647238/post-10456249)
# For the ROMs with INCREASE CARTRIDGE SPEED:
*Note: this is one of those edge cases where using a patched ROM might be needed as the SuperCard SD is NOT compatible with that feature*

* Using SuperCard team's patch tool ([Super Card V2.71](http://www.linfoxdomain.com/nintendo/ds/dl.php/supercard_microsd_v2.71_english.zip))
    * Install the patching tool (preferably via Windows XP through a type-2 hypervisor like Oracle's Virtual Box)
    * Open the ROM in the Super Card V2.71 patching tool and deselect all parameters from the ROM properties.
        * _We only need to use this tool to disable INCREASE ROM SPEED hence why we deselected every other ROM option as those patches are done automatically via SCFW._
* Ignore the status under the Patch column as that is irrelevant(doesn't matter if it shows FAIL or PASS).
* Ensure that your OUT path is somewhere you can access on your computer.
    * For this example, I'll use the "Desktop"
* Export the patched ROM by pressing the OUT button from within the Game List tab(Lower right of the window)
    * It will export 3 files:
        * {ROM NAME}.SAV
        * {ROM NAME}.SCI
        * {ROM NAME}.BIN
            * Take/Copy ONLY the bin file.
* Rename the exported ROM extension from *.BIN to *.GBA
* Transfer the ROM with the modified extension to your micro SD card.
* Play