metroid-maniac/SCFW #13: SDRAM write failed at

pitrako, 2024-06-10:
I get this message repeatedly sometimes when turning off and turning on, but If I let the console off a few seconds and turn on It goes away and saves correctly.

OmDRetro, 2024-07-08:
Issue sounds [similar to this bug](https://github.com/metroid-maniac/SCFW/issues/6).

- Can you details the steps you took to get this error?
- Please indicate the console you're using. Are you using a clone like an EXEQ Game Box or Revo K101?

pitrako, 2024-07-09:
My gba sp is AGS-001:
1-Start console and start game.
2-Turn off console.
3-Inmediately turn on console again:
[image]
4-Turn off cosole again.
5-Wait a few seconds.
6-Turn on console.
7-SCFW says is creating the save for the game.
8-Everything is normal until it happens again.

OmDRetro, 2024-07-09:
After some searching across the net, looks like there's a way to figure out if the battery cell(coin battery) has gone dry. That error indicates a dry battery based on what I found. Since the stock firmware has an SDRAM test feature on it, it actually tells you whether or not it's working as intended and a user was able to fix it by swapping out the dry battery cell with a new one.

[Reference here](https://gbatemp.net/threads/supercard-sd-micro-sd-gba-sdram-error.554029/post-9847150)

# `Solution: Try swapping out the CR2032 battery in your SuperCard SD with a new one and see if it works for you`

pitrako, 2024-07-09:
sdram is ok but flash id gives me error:
[image]
[image]
[image]
[image]
[image]

pitrako, 2024-07-10:
Now with 1.85 I get different sdram errors, flash id is the same error I think, I am going to test another battery but I don't know when, thanks:
[image]
[image]
[image]
[image]

OmDRetro, 2024-07-10:
Hm... Alright, but `since this issue is similar to another that I referenced and that both stock firmware(v1.85) and SCFW show the same problem` (SDRAM write failed).

## This is more likely tied to the edge connectors of the GBA SP not having proper contacts to the SuperCard SD pins.

### Try this first:
* Clean the pins of the SuperCard SD.
* Then try wedging in a small piece of paper under the SuperCard SD PCB to slightly elevate the board then slot it in your GBA SP.
* See if that fixes the SDRAM issue for you.

pitrako, 2024-07-14:
The pins are clean, I think is the cartridge that is bad because I have an everdrive clone that works ok, close this.

OmDRetro, 2024-07-14:
Ok, but I'd rather let you know that anything can happen especially during flashing. If the pins aren't properly touching the edge connectors, who knows what the behavior will be when you flash the cartridge?

For now, as you requested, the issue ticket will be closed. Feel free to file anew if this persists.

closed 2024-07-14 by OmDRetro