metroid-maniac/SCFW #16: fat init broken on SuperCard Lite (MicroSD)

ApacheThunder, 2024-07-01:
After finally figuring out how to compile SCFW and related projects I have investigated the issue with fat init not working with SCFW kernal on SuperCard Lite (MicroSD edition, not the Rumble version)

I have indentified the 2 commits that break fat init for this card:

https://github.com/metroid-maniac/SCFW/commit/6324930cecb0498736ba8cd3ec6f645bfb006758

And:

https://github.com/metroid-maniac/SCFW/commit/228b8b6f3abab36cf16df6261f4d987040d0ab61

The specific part of the waitstate commit that breaks SC Lite was the changing of 152043522 to 0x0b100002. I was able to keep the rest of the changes from this commit with no issue but not that final line. I had to use the old value. I tried other values but it seems to only work with the one that was originally set so I guess SC Lite doesn't like that one.

As for the "Use SWP instruction to optimize block read" commit since I don't understand thumb code very well I can't really identify why that one is not working. All I know is, the fat init fail error returns if I include that commit to the file.

I have confirmed all the other optimizations/commits work fine. It's just these two that break it. Maybe performance will still be fine without these specific commits...otherwise SC Lite will require a seperate build I suppose if you don't want to give them up for SC Lite users.

OmDRetro, 2024-07-01:
Assigning this bug to myself **temporarily**. I'll try setting up a pre-release involving your changes, but this change will be isolated to a different kernel for the time being. So for now, I'll try working on both the Supercard SD version as well as the Supercard Lite version and I'll upload a build for you to test.

OmDRetro, 2024-07-01:
This is done via [commit 40ff02f](https://github.com/metroid-maniac/SCFW/commit/40ff02f212d1c1c3dd47514647e75a2791870a4f). The latest pre-release should also have the [sclite version of the kernel](https://github.com/metroid-maniac/SCFW/releases/tag/0.5.2-HVCA-B).

Closing this bug for now. Will reopen if regressions occur.

closed 2024-07-01 by OmDRetro