metroid-maniac/SCFW #17: I got a white screen after flash the firmware

hualazimo7, 2024-07-05:
[image]

I have tried use a Ds lite to save my sc card, while it was stuck at the stage of programming 524288 bytes.

I thought the cpld ' firmware was written in the 512KB external flash, so is it broken？

Can I use a cpld programmer just programming the flash with your firmware.frm？

Actually I have replaced the 512KB flash by another one， but the .nds program seems can't   check it.

OmDRetro, 2024-07-05:
# Bricked SuperCard SD's must be restored using a Nintendo DS/DS Lite

## Method 1:

There's a build that restores the SuperCard SD back to stock firmware. You can [download this tool](https://gbatemp.net/attachments/sckill-ofw-zip.439383/) and run the `sckill-OFW.nds` on a **SLOT-1** flashcart that supports YSMenu. Just be sure to not select the SCLite option(by pressing up) before flashing the card.

### NOTE: THIS METHOD TAKES 1 HOUR AND 13 MINUTES.

## Method 2:

You can [use this](https://gbatemp.net/attachments/flashmp-zip.442627/) to restore your card though personally I have not tried this method. Instructions for this [can be found here](https://gbatemp.net/threads/scfw-custom-firmware-kernel-for-supercard.647238/post-10440565)

## Method 3:

You can [use this](https://gbatemp.net/attachments/sckiill-zip.438221/) to flash the bricked SuperCard to SCFW then restore it to OFW afterwards

OmDRetro, 2024-07-08:
Hello @hualazimo7,

If you have no further inquiries, I'll be closing this bug in 5 days as we haven't heard from you.

hualazimo7, 2024-07-08:
It works well thank you.

closed 2024-07-08 by hualazimo7