metroid-maniac/SCFW #20: White screen after firmware.frm froze on flash, unable to reflash using DS Lite

AdrianCastelloP, 2024-07-28:
Tried flashing custom firmware on a AGS-001 GBASP and it froze on me for hours, booted down, booted back up, white screen. (Important to note that I had a success on upgrading the official firmware on this cart)

I followed instrucitions on Issue #1 and Issue #17 also checked out the GBATemp thread for info, here's what I've done:

Various attempts with SCKill variations froze on 80010300 or 0x080077ff (this one I don't remember quite well if it was this address but frankly I'm tired of running these variations, point is that SCKill freezes on a certain memory address) and rendered me with the card not being recognized either by the DS Lite or GBASP

FlashMP appears to freeze, left a few times running for hours, one attempt I booted off the DS Lite and I was miraculously able to run the L+R diagnostics on Supercard, every page said OK so I assume the hardware is intact. On this same attempt the menu flashed for a brief second before turning white, tried pressing A to run a ROM, didn't work. On this same attempt, if I removed the SD Card the NO SD screen would show. Unfortunately I tried flashing afterwards and I am not able to recover back to this state.

Tried between SCKill and FlashMP several other times, with both of them freezing on me, left them running for hours, but rendering either card not being recognized by both my consoles or not being recognized at all.

Official SuperCard recovery .nds does not work on my R4 flashcard, it is the only option that I was unable to attempt, if there is any instructions on how to get it running I will try it.

Below is my SuperCard Model:
[image]
[image]

OmDRetro, 2024-08-13:
You can try using the [FlashMP](https://gbatemp.net/attachments/flashmp-rar.345540/) method via your NDS/NDSL.

If that still fails, your flash chip has gone bad. Normally if flashing stops at a certain address it indicates a failure in flash memory which would require replacement of the chip marked U1 [image]

* Image referenced from [davidgf's site](https://www.davidgf.net/2024/07/24/arm-emulation-static-analysis/)

OmDRetro, 2024-08-19:
Lucky you, I experienced the same thing with one of my 4 SuperCard SD's and my hunch is correct. This is a __HARDWARE LEVEL PROBLEM__ and __IT ISN'T__ a __KERNEL / SOFTWARE PROBLEM__

My observations are correct. The flash chip has gone bad since at the last of the flash chip's life I was able to flash this back to OFW, but the text is full of garbage. (unreadable). Flashing this back to custom firmware just leads to what you have experienced.

Since __HARDWARE LEVEL__ concerns are __NOT PART OF THE PROJECT'S SCOPE(SOFTWARE)__, you may need to get someone who's capable of removing the flash chip and soldering in a new one.

Closing this "bug" as irrelevant.

closed 2024-08-19 by OmDRetro