metroid-maniac/SCFW #22: Unable to use the MPA and txt

SomeAbyss-One, 2024-08-06:
Hello and sorry for the another question but I've been having trouble getting an txt and mpa file to work. Firstly, I don't know which e-book viewer to use as all the ones I've try have either blank screened or didn't open the file and have only been showing sample text. Secondly, I've converted a mp3 using the instructions provided into a mpa file though when trying to run it with Musicplayer advance by renaming it to mpa.gba in scfw it gave a whitescreen. so

OmDRetro, 2024-08-07:
The instructions are in the [README.md](https://github.com/metroid-maniac/SCFW/tree/OmDRetro-SCFW-Kernel). You skipped a step requiring the ⚠ `gbafix` ⚠ step(this step makes use of devkitpro's bundled tools).

On the other hand, you can also just download a copy of all the compiled emulators / apps from this [GBATemp post](https://gbatemp.net/threads/scfw-bleeding-edge-modular-kernel-branch.656629/).

SomeAbyss-One, 2024-08-07:
I'm having trouble running the tool kit on both mac and windows. Is there another alternative or pre patched file ?

OmDRetro, 2024-08-11:
@SomeAbyss-One : I answered your inquiry so I'll be closing this after a day if you do not respond.

closed 2024-08-12 by OmDRetro