metroid-maniac/SCFW #23: Revert firmware patch

fernandobonfimandrade, 2024-08-14:
I patched the supercard's firmware with SCFW.frm, but the games I already had on the card no longer run. Do I need to put the roms without the supercard software patch? Is there a possibility to go back to the original firmware?

OmDRetro, 2024-08-16:
TL;DR: Yes, by default you shouldn't need the patcher, but if you want to use the patched binaries you'll need to disable the patching via kernel settings and rename the ROM extension from `*BIN` to `*GBA`. There are [special cases](https://github.com/metroid-maniac/SCFW/issues/10) where you need to use the patcher though.

OmDRetro, 2024-08-19:
@fernandobonfimandrade , If you have any other questions, please write them down. Otherwise, I'll close this after 4 days.

closed 2024-08-23 by OmDRetro