metroid-maniac/SCFW #25: Contributing to the project

sideprojectslab, 2024-09-02:
Hello, I like what you are doing and I would like to help out. I am involved in a few retro projects myself and have been waiting for a very long time to pick up support for the supercard. Is there a way we can get in touch to go over if and how I could lend a hand? Thanks!

sideprojectslab, 2024-09-15:
The project is abandonware innit?

metroid-maniac, 2024-09-15:
I haven't had the time or energy to maintain this project for a while. Did you have anything in mind that you wanted to add?

sideprojectslab, 2024-09-15:
Heheh, sorry for the little poke there. In general I wish for this project to move on, as I think it is really cool, and I would be willing to put in the work. In general I would like to work on the "patching" side of things. I see for instance that one still needs to patch FFVI (at least partially) with the original supercard software for it to work with scfw. There are two sides to this. On one hand I would like to take what you have learned about ROM patching and turn it into an "offline" PC patcher as a replacement for the supercard software. After that, I would like to work on the "on the fly" ROM patching that you have implemented in scfw to try and make it faster and improve ROM compatibility (as I found several games that launch to a white screen with scfw). For this I would need some guidance and a somewhat efficient communication channel with you or other contributors of scfw, so I can pick your brain as I chip away at the problem, what do you think?

OmDRetro, 2024-09-15:
We actually do need help with some features and thanks a lot for reaching out.

On my end, I'm still working on a few things that still need to be added like the full implementation of HVCA instead of the stripped down implementation that it currently is at the moment. This will allow users to toggle emu priority between PocketNES and HVCA for NES ROM loading...

We do have a pretty barebones discord server, but it would be best to send the invite link over GBA temp(I believe you're also there?).

OmDRetro, 2024-09-15:
### Contribution request

Closing this bug/ticket as this is not relevant to a kernel/firmware problem.

closed 2024-09-15 by OmDRetro