metroid-maniac/SCFW #26: Problem starting gba games

lorenzogamingtus, 2024-09-18:
Hello, my GBA games don't work, they launch but make a shrill noise or freeze on (Nintendo Game Boy), except Mario and Luigi Superstar Saga, thank you

OmDRetro, 2024-10-01:
### Insufficient information

Please provide the ff.:

* Are you on a `Release` kernel or `Pre-release` kernel?
     * Are you using the kernel on a:
          * `SuperCard lite micro SD`
               * [image]
          * `SuperCard Mini SD / Micro SD`
               * [image]
          * `SuperCard SD`
               * [image]
* Which game are you trying to load?

TsilaAllaoui, 2024-10-01:
Hi, me too. I use original AGB-001 with modded screen and only get white screen after loading game or infinite loop on loading kernel. I use Release version and SuperCard SD Phat one

lorenzogamingtus, 2024-10-01:
hello thank you for answering me, I have a supercard micro sd in kernel release and I try Mario and Luigi superstar saga and the pokemon emerald and Metroid fusion moreover I have an error: flash id error: 00b9 0000c2ba Thank you

TsilaAllaoui, 2024-10-01:
@OmDRetro Is there some manipulation to do to make it work on AGB-001 model? I saw like booting in multiboot mode for 4 sec then loading  game but it doesn't work for me. Note that it works on SP, Micro and DS Lite

TsilaAllaoui, 2024-10-01:
@OmDRetro Is this pattern valable for every game?

OmDRetro, 2024-10-01:
> @OmDRetro Is this pattern valable for every game?

Unlikely, but not impossible. If you're looking for an exhaustive list, you wouldn't find it here.

Sounds like you're allergic to the `SuperCard 2.71 patcher` 😅

I'd suggest to follow the instructions shown in [Issue#10](https://github.com/metroid-maniac/SCFW/issues/10)
* [Refer to this](https://github.com/metroid-maniac/SCFW/issues/10#issuecomment-2266148248)

TsilaAllaoui, 2024-10-01:
@OmDRetro Yeah, i'm not a big fan of the patcher 😅 any way to use the kernel to flash the rom? or another homebrew to do this? I can contribute if there is a lead you can show me

OmDRetro, 2024-10-05:
@lorenzogamingtus if you don't have any other concerns for this particular issue you have, I'll be closing this after 3 more days.
***
@TsilaAllaoui The hex pattern is the result of using the `SuperCard SD patcher 2.71` application. I think you made it clear that you don't want to use it, but I haven't experimented with other similar patchers out there to provide you with more assistance. As a team, we're already aware of the kernel's limitation as of late but implementing the patches to disable `Increase ROM speed` may take some time. On a personal note, I'd rather call it `Regulate ROM speed` by the time its implemented, but I'm not in charge of implementing that part of the patching system as of this time.

> [!IMPORTANT]
> **Do not expect an ETA of when this feature will be done. It will be uploaded as soon as it's ready**

closed 2024-10-05 by lorenzogamingtus