metroid-maniac/SCFW #28: AGB-001 Can't load rom (White screen)

TsilaAllaoui, 2024-10-01:
Hi, i use a model AGB-001 console and two cases appear:
- It fail on loading kernel at start 
- It fail on booting the rom, freeze in white screen (any rom)

Note that it works on GBA SP, Micro and NSD Lite model. 
I use:
- SuperCard SD (Phat older model) 
- Release kernel

OmDRetro, 2024-11-03:
Hi @TsilaAllaoui,

There are 2 models of the AGB-001:
* 40-pin LCD model (Later revision)
* 32-pin LCD model (Earliest/Initial release AGB-001)

Found out about the two AGB-001 models here:

[image]

Please indicate which one you have. Just letting you know that I have the 40-pin LCD model AGB-001

TsilaAllaoui, 2024-11-03:
I have a 40pin gba AGB-001 with modded screen from an AGS-101

OmDRetro, 2024-11-30:
I'm not able to replicate this problem on my GBA AGB-001(Same model as yours. 40-pin also). Have you tried checking if the pins of the flashcart is making proper contacts with your GBA's edge connector? [Some users on reddit](https://www.reddit.com/r/Gameboy/comments/7ci0sm/fixing_bricked_supercard_mini_sd/?sort=new) reported that wedging in a piece of paper on the underside of the board elevating it enough to have proper contact with all the GBA pins.

OmDRetro, 2025-02-24:
I, personally, was not able to reproduce this issue. But there have been reports indicating [similar behavior for other custom firmwares(SuperFw) for this flashcart](https://gbatemp.net/threads/superfw-a-beta-custom-firmware-for-supercard.654847/post-10593513)

> [!NOTE]
> All sorts of unexpected behaviors can occur when the GBA's edge connectors have no proper contacts with the cartridge pins as [detailed here](https://www.reddit.com/r/Gameboy/comments/7ci0sm/fixing_bricked_supercard_mini_sd/?sort=new)

As of now, I'll be closing this as `NOT REPRODUCIBLE`.

closed 2025-02-24 by OmDRetro