metroid-maniac/SCFW #29: GB save filename issues

jinyukyo, 2024-10-31:
the last char of the GB save filename is missing

OmDRetro, 2024-10-31:
Sounds like the filename was truncated due to it being longer than the usual which is expected if the filename is way too long.

Since SFCW version 0.2, `the best practice is to ensure that the ROM filename is at a reasonable length` which is more or less 32 characters much like how most GBA emus handle ROM files.

- Did the long filename prevent you from loading your save?
    - If yes:
        - How long is the filename of the ROM your playing?
        - How many subdirectories is the ROM in?
        - Have you tried trimming the filename?
    - Otherwise, it's not an issue

jinyukyo, 2024-11-01:
[image]
sorry my bad english,gb savefile name is different from gb rom name,but gbc is correct.

jinyukyo, 2024-11-01:
sorry my bad english,gb savefile name is different from gb rom name,but gbc is correct.
[image]

OmDRetro, 2024-11-02:
> sorry my bad english,gb savefile name is different from gb rom name,but gbc is correct. ![savename](https://private-user-images.githubusercontent.com/5707099/382237423-f5fa78d7-db22-486b-8627-d627a02df699.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzA1Mjc3NTgsIm5iZiI6MTczMDUyNzQ1OCwicGF0aCI6Ii81NzA3MDk5LzM4MjIzNzQyMy1mNWZhNzhkNy1kYjIyLTQ4NmItODYyNy1kNjI3YTAyZGY2OTkucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI0MTEwMiUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNDExMDJUMDYwNDE4WiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9NWMzZjkxN2U4M2MwODRkYzE1N2ZhMTkzOWY2ODUxZmU4Mjk4ZjViOTE4YTc0ZjY1NGYzYzhkZDA5MWYwZjk3YiZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QifQ.itAPrJwfzSDW10z3R4XobhTxRMT95TV0roVMbtlLY9o)

Actually, good catch.
This is an oversight on my part, but it is relatively minor as the saves still load for the OG GB games
I'll fix this on my next build

OmDRetro, 2024-11-02:
Thanks for notifying me about that oversight, @jinyukyo .

Just letting you know that this is fixed and it will be included in the next build. 

I'll be marking this as `resolved` once I push the changes.

[image]

OmDRetro, 2024-11-03:
This is fixed in commit [d264d14](https://github.com/metroid-maniac/SCFW/commit/d264d14ed1fc06f7042a15d0f6c01bb78378c7e5)

closed 2024-11-03 by OmDRetro