metroid-maniac/SCFW #30: How on earth do I apply a gbafix to the emulators before putting them on the flash cart (this is the only place I could find to ask)

44cK3r, 2024-11-05:
I don't do forums sorry so I will never ask there sorry

OmDRetro, 2024-11-05:
There's a copy of `GBA fix` in [this repository](https://github.com/Sterophonick/mirror-meteo-avi2gba) without having to download all the toolchains provided by `devkitPro`.

- Simply drag and drop the binary on top of the `gbafix.exe` binary. That should patch the plugin/ROM, hence you "GBA fixed" it.
***
I know it's a bit tedious for the normal user to prepare all of the prerequisites by themselves, but I always collate the entire package(kernel + firmware + plugins) into a zip file in [this GBATemp thread](https://gbatemp.net/threads/scfw-bleeding-edge-modular-kernel-branch.656629/). Just keep an eye out and download from there instead so you don't have to do any of these extra steps.

- Download your respective kernel package
- Extract the contents to the root of your SDCard and all the plugins should work.
    
>[!NOTE]
>I'll be marking this issue as resolved once I upload the package to the aforementioned thread.

OmDRetro, 2024-11-05:
This is done. Please note that `Famicom Disk System` games will not work without the FDS bios. It is not included in the package.

closed 2024-11-05 by OmDRetro

44cK3r, 2024-11-05:
YAY thank you so very much

44cK3r, 2024-11-26:
> This is done. Please note that `Famicom Disk System` games will not work without the FDS bios. It is not included in the package.

sorry to bother you but I'm just wondering why the forum hasn't gotten the last 2 updates in complete compiled packages