metroid-maniac/SCFW #32: EBook Advance crashes on startup

dopeyhex, 2024-11-22:
I'm using the beta1.1 verison of txt.gba. If you try to read any .txt files, glitched out splash screen will appear with the broken pink logo and console will freeze.

OmDRetro, 2024-11-23:
## Beta version 1.1 does not exist.
I already linked the latest version in the project's [README.md for OmDRetro-SCFW-Kernel](https://github.com/metroid-maniac/SCFW/tree/OmDRetro-SCFW-Kernel)
***
Daniel C., the author, has not made a version beyond v0.3.
Where did you get that version?

OmDRetro, 2024-11-26:
[dopeyhex](https://github.com/dopeyhex) I'll be closing this in 3 days if you do not respond to this.

dopeyhex, 2024-11-27:
Oops, I meant to say 0.11 beta Final from the page https://web.archive.org/web/20070512224717/http://members.optushome.com.au:80/dancotter/ebook.htm

OmDRetro, 2024-11-30:
Be sure to read the `readme.txt` file. It should have some pointers on how to properly use the ebook reader. I know there's a prerequisite requiring the use of the `wrap.exe` first in order for the `*.txt` files to be read properly on `EBook Advance`. Otherwise, you'll run into some problems. This GBA application likely isn't in a stable state hence the version being `0.3`.

Since you've not any issues relating to SCFW, I'll be closing this now. Feel free to reopen if you find any specific SCFW loading problem.

closed 2024-11-30 by OmDRetro