metroid-maniac/SCFW #33: some game soft reset patch doesn't  working..

jinyukyo, 2024-11-26:
Why do soft reset patches fail to work in some games, such as 《Apotris》?
Additionally, the button combination for The Legend of Zelda is also A + B + Start + Select.
So, the game cannot save properly due to the conflict with the soft reset button combination.

OmDRetro, 2024-11-27:
@jinyukyo , you have two concerns here:

* Soft reset patching not working on some games
* Soft reset key combination conflicts with a GBC game(Goomba)

***

### Soft reset key combination conflict

- All you need to do is to disable the `Soft reset Patch` by toggling it in the kernel settings. The key value should be `0`.
    - Once disabled, you won't be able to soft reset back to the kernel menu but you will be able to quit to kernel.
- Alternatively, if you want to keep the soft reset, you can make use of `Goomba Color's save state feature`.

OmDRetro, 2024-11-30:
### Soft reset is unsupported by some ROMs OR requires manual intervention to patch the ROM
- If you have the expertise to patch the ROM with soft reset, you may have to do that first before adding the ROM to your SD card.
- Alternatively, if SCFW cannot patch the ROM directly with soft reset, you may try using [Boot-O-Mat Adance](https://www.gamebrew.org/wiki/Boot-o-Mat_Advance_GBA) instead to do this for you.

Also, I'll be closing this in 4 days as I've already answered your inquiry.

OmDRetro, 2024-12-07:
Closing this as not planned since some ROMs will require manual intervention to patch them and that assumes you have the knowledge to be able to patch the ROM yourself.

closed 2024-12-07 by OmDRetro