metroid-maniac/SCFW #34: 程序编译报错，在链接阶段缺少相关的函数

ForgotFun, 2025-03-10:
forgotfun@DELL-T3240:~/project/gba/SCFW/SCFW_Kernel_GBA$ make
main.c
/home/forgotfun/project/gba/SCFW/SCFW_Kernel_GBA/source/main.c: In function 'sort_folder_nickname':
/home/forgotfun/project/gba/SCFW/SCFW_Kernel_GBA/source/main.c:68:34: warning: initialization discards 'const' qualifier from pointer target type [-Wdiscarded-qualifiers]
   68 |         struct dirent_brief *l = lv;
      |                                  ^~
/home/forgotfun/project/gba/SCFW/SCFW_Kernel_GBA/source/main.c:69:34: warning: initialization discards 'const' qualifier from pointer target type [-Wdiscarded-qualifiers]
   69 |         struct dirent_brief *r = rv;
      |                                  ^~
my_io_sc_common.c
my_io_scsd.c
my_io_sd_common.c
overclock_ewram.c
Save.c
WhiteScreenPatch.c
my_io_scsd_s.s
linking cartridge
/opt/devkitpro/devkitARM/lib/gcc/arm-none-eabi/14.2.0/../../../../arm-none-eabi/bin/ld: main.o: in function `resetPatch':
/home/forgotfun/project/gba/SCFW/SCFW_Kernel_GBA/source/main.c:282:(.text+0x574): undefined reference to `irq_hook_bin_len'
/opt/devkitpro/devkitARM/lib/gcc/arm-none-eabi/14.2.0/../../../../arm-none-eabi/bin/ld: /home/forgotfun/project/gba/SCFW/SCFW_Kernel_GBA/source/main.c:282:(.text+0x590): undefined reference to `irq_hook_bin'
/opt/devkitpro/devkitARM/lib/gcc/arm-none-eabi/14.2.0/../../../../arm-none-eabi/bin/ld: /opt/devkitpro/libgba/lib/libfat.a(disc.o):(.rodata+0x4): undefined reference to `discGetInterface'
collect2: error: ld returned 1 exit status
make[1]: *** [/opt/devkitpro/devkitARM/gba_rules:40: /home/forgotfun/project/gba/SCFW/SCFW_Kernel_GBA/SCFW_Kernel_GBA.elf] Error 1
make: *** [Makefile:111: build] Error 2

因为代码中提供的是win下的工具，所以使用的交叉编译工具链是devkitpro install脚本安装的，libgba和libfat是编译安装的。discGetInterface irq_hook_bin irq_hook_bin_len 这几个函数未定义，所以链接报错。

ForgotFun, 2025-03-10:
Because the code includes tools designed for Windows, the cross-compilation toolchain was installed using the devkitpro install script. Both libgba and libfat were built and installed from source. However, the functions discGetInterface, irq_hook_bin, and irq_hook_bin_len are undefined, which causes linking errors during compilation.

OmDRetro, 2025-03-22:
@ForgotFun : The current versions of devkitARM do not support the SuperCard library as it was deprecated on releases after r38. This means the project cannot build on current versions.

You need to use an older release(`DevkitARM_r38`) which is included in this repository.

Afterwards, run the __`MSYS`__ shell and set the environment variables of both DEVKITPRO and DEVKITARM to the right directories.

Sample:
* The commands below assume that your __`devkitARM_r38`__ is found in the root of the C drive
```
export DEVKITPRO=/c/devkitPro/
export DEVKITARM=/c/devkitPro/devkitARM
```

> [!NOTE]
> Exporting environment variables is temporary for the session you're on. 

Alternatively, you can also set the environment variables:

```
set PATH=C:\devkitPro_gba\msys\bin;C:\devkitPro_gba\devkitARM\bin;%PATH%
set DEVKITARM=/c/devkitPro_gba/devkitARM
set DEVKITPRO=/c/devkitPro_gba
set LIBGBA=/c/devkitPro_gba/libgba
```

Proceed to build the project and you're good to go

OmDRetro, 2025-04-01:
Hello @ForgotFun ,

I'll be closing this "issue" since I haven't heard from you. Feel free to reopen though

closed 2025-04-01 by OmDRetro